johncarlson21/SV04-Marlin-2.1.x #3: Changing flow on first layer skips part of the queue

lunatic1972, 2023-02-17:
Initially I had my flow turned down intentionally in prusa slicer(V. 2.5.0 released) to around 96%(0.96), with a minimum of 4 loops. I seen how the first loop progressed and while it was starting the second loop, I adjusted the flow on the screen to 102% on Nozzle 1. It then proceeded to move the nozzle to start the actual print without completing the rest of the loops. Print completed but it skipped part of the queue in marlin with changing the flow.

johncarlson21, 2023-03-04:
I'm not sure I understand what you're saying here.. I get what you're saying about changing it on the screen, I'm maybe not understanding what you mean about the loops.

lunatic1972, 2023-03-09:
After I reuploaded the firmware apparently the problem went away. You can close this one

johncarlson21, 2023-03-09:
That is great :)

closed 2023-03-09 by johncarlson21