johncarlson21/SV04-Marlin-2.1.x #4: [Wrong extruder] Uses wrong extruder in single mode

rainey832, 2023-02-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

 If you print something in single mode and then immediately print that same thing again without turning it off it decides to use the second extruder.

### Bug Timeline

New

### Expected behavior

Expected to use the selected extruder in single mode whenever which extruder is used isn't specified in starting gcode

### Actual behavior

Printer will use the selected extruder once, but if that print completes and you immediately start the same print with the same extruder selected again it will alternate to the opposite extruder. In this case it alternated to extruder 2. This is fixed by power cycling. Then you can select whichever extruder and it will use the expected extruder.

### Steps to Reproduce

1. Slice any STL and do not specify which extruder its to use in single mode
2. Print file, and then immediately print it again with the same extruder selected in single mode.
3. Printer then will alternate to the secondary extruder.

### Version of Marlin Firmware

2.1.2

### Printer model

Sovol SV04

### Electronics

Stock

### Add-ons

Stock

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

johncarlson21, 2023-02-19:
Please upload your gcode file you are testing with so I can see it.. I just sliced using cura and removed the calls to the extruder to set it, and it seems to be working for me. I will post a video to your FB comment.

Also you should, never NOT set the extruder and let the printer guess.

rainey832, 2023-02-19:
I've attached the gcode file. Thank you for time with this. So I was under the impression since I can choose which extruder to use in the mode selection touchscreen, I could take a single extruded gcode file and pick on the printer which extruder it uses. If i'm to always set the extruder to T0 for a single extruded file for example, would I still be able to select extruder 2 on the printer itself and have it use extruder 2?

rainey832, 2023-02-19:
[LIGHTSABER-HILT_BLACK_12h50m_LH0.16mm_N0.6mm.Zip](https://github.com/johncarlson21/SV04-Marlin-2.1.x/files/10777256/LIGHTSABER-HILT_BLACK_12h50m_LH0.16mm_N0.6mm.Zip)

Sorry had to zip

rainey832, 2023-02-19:
Sorry again, I sent a dual extrude file
[OBI-WAN-LIGHTSABER-CAP_1h1m_LH0.16mm_N0.6mm.zip](https://github.com/johncarlson21/SV04-Marlin-2.1.x/files/10777266/OBI-WAN-LIGHTSABER-CAP_1h1m_LH0.16mm_N0.6mm.zip)

johncarlson21, 2023-02-19:
Yep your end code is setting T1.
Have a look at this pic.. Line 144437
[image]

johncarlson21, 2023-02-19:
Need to make sure you are using the correct slicing printer profile for this or check your printer end code..
but my suggestion is to let the slicer define the extruder.

rainey832, 2023-02-19:
Wow that is fascinating thanks, I didn't realize that would carry to the next print. Alright so not a bug. Thanks again

closed 2023-02-19 by rainey832