johncarlson21/SV04-Marlin-2.1.x #5: Individual fan speeds can't be controlled with M106/M107

lunatic1972, 2023-02-20:
I have some custom Tool Change Code in prusa slicer where for every tool change it does the following
{ if previous_extruder>-1 } 
T{previous_extruder}
M106 S0
T{next_extruder}
M106 S255
{ endif }

What happens is the previous tool parks as expected, turns the fans off very briefly and then the other tool head moves into position and both fans turn on. I've verified the gcode actually expands to the correct T0/T1 as expected but the unused one still has full fan speed. Through the touch screen however you can control it ON/OFF

johncarlson21, 2023-02-28:
You need to tell it what fan..
The way it kinda works now, is that it just turns both of them on.. if you want in your gcode to do this correctly, you need to set it like this:
M106 P0 S255 ( extruder 1 )
M106 P1 S255 ( extruder 2 )

I believe this is a type of safety feature for dual head printing. Like for Mirror/Copy mode, the machine has no idea that the second head is doing anything, so it turns on both fans.

lunatic1972, 2023-03-01:
Thanks for that. You can probably close this out then. I haven't tried it out yet but makes sense until Marlin has better control with IDEX in the 2.1.x branch

lunatic1972, 2023-03-01:
Well, I was going through the code a bit for M106_107 and changed some stuff, tried it out and it worked. I haven't tried it in duplication/mirror mode but will try it soon.  Here is the code
Basically when the non active tool is parked, the fan is off otherwise the fan is on as normal. Crude but works

#if ENABLED(DUAL_X_CARRIAGE)
    if (dxc_is_parked())
    {
      if (pfan==0 && !active_extruder_parked)       //RBB added from here
        {
          thermalManager.set_fan_speed(1, speed);
          thermalManager.set_fan_speed(0, 0);
        }
      if (pfan==1 && !active_extruder_parked)
        {
          thermalManager.set_fan_speed(0, speed);
          thermalManager.set_fan_speed(1, 0);
        }                                           //RBB Added to here
      
      //thermalManager.set_fan_speed(0, speed);     //JC Original code
      //thermalManager.set_fan_speed(1, speed);     //JC Original code
    }
  #endif
  TERN_(LASER_SYNCHRONOUS_M106_M107, planner.buffer_sync_block(BLOCK_BIT_SYNC_FANS));

  if (TERN0(DUAL_X_CARRIAGE, idex_is_duplicating()))  // pfan == 0 when duplicating
	{
		thermalManager.set_fan_speed(0, speed);
		thermalManager.set_fan_speed(1, speed);
	}

johncarlson21, 2023-03-04:
I will have to look at this sorry it is taking so long I have been super busy