johncarlson21/SV04-Marlin-2.1.x #6: [BUG] Parking head with e.g. M125 leads to head crash

endrift, 2023-03-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When using a G-Code like M125 to pause printing, it will try to park the head all the way to the right...which leads to head 1 crashing into head 2, resulting in a layer shift. I believe this is due to NOZZLE_PARK_POINT using Y_MAX_POS, but I'm not sure--I haven't tested it. In the meantime I'm gonna be setting the park point manually, I suppose.

### Version of Marlin Firmware

V2.1.3

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I used your release so the configuration is as presented...

cryoglobe, 2023-03-11:
I think I got a similar case , but after a power failure while printing in copy mode ... 

After restarting the machine , heating , 
Head 2 was going in its position ( as in the beginning of copy mode ) , but suddenly 
both head moved to the extreme right , so E2 crashed when trying to go too far , and E1 crashed on E2
Then both heads moved back to the print place , but none in the correct position 

Conclusion : print is failed , need to switch off the printer.

johncarlson21, 2023-03-18:
Yes there is an issue with the nozzle park point. 

I have this as a fix to be pushed up soon. 

There is also a different park point for pause vs advanced pause filament change which I'm looking into.

johncarlson21, 2023-03-30:
I've updated the nozzle park point in version 2.1.3.1

endrift, 2023-03-31:
Does the updated park point apply to both heads the same? Or are there unique park points per head?

closed 2023-04-05 by johncarlson21