johncarlson21/SV04-Marlin-2.1.x #7: Esteps

Piccyman, 2023-03-17:
**On the esteps adjustment, there should be x and y and z for each extruder.
Lovely add-on

johncarlson21, 2023-03-18:
I looked into this for another machine I built and you cannot set the esteps independently for x or y

I've only put this in for the extruder motor

I believe you can set for the z but have not looked into this.

johncarlson21, 2023-04-03:
I will be adding X, Y, Z steps to the next update of the screen and the firmware.

Piccyman, 2023-04-03:
Thank you, I did ask a question on Facebook, is linear advance enabled ?

johncarlson21, 2023-04-03:
Yes it is, and it does have some default set.
I'm not really familiar with linear advance etc.. I know if you want to get it just right you need to do some testing.

johncarlson21, 2023-04-05:
This is now updated in the V2.1.3.1b3 pre-release. You also need to update your screen to 1.2.7
Linear Advance is enabled and I've change the K value to 0 so it is basically disabled. You can use a PC to update your values for K.

closed 2023-04-05 by johncarlson21