johncarlson21/SV04-Marlin-2.1.x #8: Z offset

Piccyman, 2023-03-17:
The z offset indicates .001mm + or -, but the movement when pressed is 0.1mm, i measured it with feeler gauges.
[image]

johncarlson21, 2023-03-18:
What version of the printer firmware are you on and what version of the screen firmware?

johncarlson21, 2023-03-18:
You are correct, I found that the increment number was wrong.. I am adding this to the next fix.

johncarlson21, 2023-03-30:
I've fixed this in 2.1.3.1b

johncarlson21, 2023-03-30:
actually I think I need to fix something in the display also

johncarlson21, 2023-04-03:
I've added a fix to the display also.

johncarlson21, 2023-04-05:
This is now pushed so you can update your screen to V1.2.7 and also update your printer firmware, I would suggest updating it to the pre-release V2.1.3.1b3

closed 2023-04-05 by johncarlson21