johncarlson21/SV04-Marlin-2.1.x #9: Manual movement limited.

lokertr, 2023-03-18:
I have the CXC from Ember Prototypes. In order to use it, both tool heads need to be able to move over the same part of the bed. Currently the two heads are prevented from doing that, at least on the touch screen. This occurs despite the fact the inactive tool head is parked off the bed. In a future update if you could set up the manual movement to be able to move both tool heads to any X position on the bed, it would fix the issue. Even just allowing a small overlap in the middle would allow the printer to work with this device.

johncarlson21, 2023-03-30:
This is because the printer wasn't homed.. you need to push the home button if this happens. also need to do that if you switch to nozzle 2

lokertr, 2023-04-01:
That works for me. Thanks.

closed 2023-04-01 by lokertr