johncarlson21/SV04-Marlin-2.1.x #10: More a doubt that an issue!

giulio-alfieri, 2023-03-20:
First of all, congratulation on the EXCELLENT job. My SV04 prints like a charm and the DWIN display is beautiful.
I´ve made just some cosmetics adjustments:
- Enabling the PID individual adjustments
- define NUM_RUNOUT_SENSORS   1          // Number of sensors, up to one per extruder.
- define TOOLCHANGE_ZRAISE                 2 // (mm)
However, I wonder why you didn´t use the DRIVER_TYPE  TMC2209_STANDALONE instead of the 4988. Any known problem?

johncarlson21, 2023-03-30:
Hello and thanks for checking..

PID individual is fixed in 2.1.3.1b and the NUM_RUNOUT_SENSORS was also updated.

As for the Z raise, I'm not a fan of raising Z and blobbing. There is also another spot where I set min z raise to 0.

As for the drivers, this is what was used in the Sovol firmware and I haven't tested changing this yet. Planning on this at some point.

giulio-alfieri, 2023-03-30:
For the Driver type, I put the TMC2209_TSANDALONE, and it is working like a charm... Kudos again for the excellent job.

johncarlson21, 2023-04-03:
The drivers will be updated in the next update

giulio-alfieri, 2023-04-03:
John,

What did you do leading up to this? Did you just start a print and then go
to the tune menu? Did you click to adjust flow or fan?

               As you can see here
https://drive.google.com/file/d/1wv5356JGd39VwZlou7YnApOY86Ck4cUP/view?usp=sharing
I've
started the print and touch Tune, and its freezes before any other command.

              After the print finished, the screen changed to the finish
screen, see here
https://drive.google.com/file/d/1e44OlHYDwKZmFF8DutanTNwHXFJUZHLu/view?usp=sharing
,
but froze for the touch.

Can you show me a screenshot of your Printer Info Screen? Just so I can
make sure that the screen versions updated.

               here the screen of printer info.
https://drive.google.com/file/d/1y6DDTByG4t9SgQDLmO63FPWaWAfQDBQ4/view?usp=sharing

*Giulio C. Alfieri*
+55 (11) 99914.1768

"O pessimista reclama do vento; o otimista espera que ele mude; o
sábio ajusta as velas."
John Maxwell

closed 2023-04-04 by giulio-alfieri