johncarlson21/SV04-Marlin-2.1.x #11: SV04 doesn't know the print is finished when printing via Octoprint

naskinner999, 2023-03-24:
**Describe the bug**
Printing via Octoprint, at the end of the print, the SV04 doesn't seem to recognise that the print is finished and the screen remains on the printing screen with the buttons for "Stop", Pause" and "Tune", rather than reverting to the "Home" screen. Requiring the user to press the "Stop" button and then confirm to return to the "Home" screen

**To Reproduce**
Steps to reproduce the behavior:
Run a print via Octoprint and wait until the print is completed.

**Expected behavior**
At the end of the print, the SV04 should revert to the home screen without user intervention

johncarlson21, 2023-04-03:
actually it would go to the finish screen, where there is a finish button.
I want customers to be able to come back and know the print is done.
I'm not sure why it isn't showing you this at the end of the print.
I will look at the code to see if I can find something.

[image]

johncarlson21, 2023-04-03:
Yeah without making a new gcode function, there is no way for the firmware to know that you are done sending code.
So I'm not sure I will be adding this.
However you could try adding M77 at the end of your gcode file, which should trigger stop print job timer and show the finish print screen.

naskinner999, 2023-04-03:
Hi John,

Thanks for looking at this.

I can confirm that adding M77 into the End G Code does fix this, though I'm not really sure why it should be necessary:

Looking at the Marlin G Code reference for M77, it says

 _"When PRINTJOB_TIMER_AUTOSTART  is enabled the print job timer  will be stopped in response to the heaters being turned off."_ 

PRINTJOB_TIMER_AUTOSTART is uncommented in configuration.h in your firmware, so I assume it is active. In this case, I wouldn't have thought it would be necessary to add M77 to the end G Code.

In fact, I had considered using M77 myself, but when I looked at your source code, I assumed the timer would be stopped automatically. Well we know what they say about assuming..!

Anyway, the main thing is it works..!

Thanks again.

johncarlson21, 2023-04-03:
The problem is that it needs to send to the screen, and this happens in the M77 and when the queue is running and it sees that it has reached end of the card file.
I really don't want to put it into another spot when these both work for screen and M77 is the option for the host.
Yes the print job timer is stopped, however there is nothing there to send a screen change.

naskinner999, 2023-04-03:
Ah, that explains it...
It's no hassle to add M77 in the End G Code. 
If that's the easiest way to resolve this, there's absolutely no need to look at changing the firmware.

Thanks again...

closed 2023-04-03 by johncarlson21