johncarlson21/SV04-Marlin-2.1.x #12: Screen frezzing

giulio-alfieri, 2023-04-03:
During a print, the Tune screen freezes, and the upper portion doesn´t recognize the touches.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the Main screen during printing.
2. Click on the Tune button
3. Click to return or Z offset
4. See, the screen doesn´t recognize the commands.

**Expected behavior**
Normal button commands have been recognized.

**Screenshots**
See the movie at https://drive.google.com/file/d/1wthAncXLiA_4KpMk4Q1GJKuJWQ5VYgX6/view?usp=sharing

**Desktop (please complete the following information):**
 - OS: W11
 - Browser Chrome
 - Version V2.1.3.1b2
 -  Screen Version 1.2.6 

**Additional context**
Add any other context about the problem here.

johncarlson21, 2023-04-03:
What did you do leading up to this? did you just start a print and then go to the tune menu? Did you click to adjust flow or fan?
Can you show me a screenshot of your Printer Info Screen. Just so I can make sure that the screen versions updated.

giulio-alfieri, 2023-04-04:
John, I sent the answer by reply the e-mail but I think something got wrong. here is the answer.

What did you do leading up to this? Did you just start a print and then go to the tune menu? Did you click to adjust flow or fan?

               As you can see here https://drive.google.com/file/d/1wv5356JGd39VwZlou7YnApOY86Ck4cUP/view?usp=sharing I've started the print and touch Tune, and its freezes before any other command.

              After the print finished, the screen changed to the finish screen, see here https://drive.google.com/file/d/1e44OlHYDwKZmFF8DutanTNwHXFJUZHLu/view?usp=sharing , but froze for the touch.

Can you show me a screenshot of your Printer Info Screen? Just so I can make sure that the screen versions updated.

               here the screen of printer info. https://drive.google.com/file/d/1y6DDTByG4t9SgQDLmO63FPWaWAfQDBQ4/view?usp=sharing
Giulio

johncarlson21, 2023-04-04:
I seen the videos..
I would suggest to write down your settings, offsets, z offset and esteps and re-flash the printer firmware.
Then go to hotend offset screen and hit the initialize eeprom button, go back in to hotend offset and hit save, just to be sure.
Then re-enter in your offsets and settings. restart printer.
I would also reflash the screen.
I'm thinking it is just the initialize eeprom but just to be sure.

Did you do the initialize eeprom when you installed the firmware?

giulio-alfieri, 2023-04-04:
John fixed the issue by reflashing the firmware and initializing the EEPROM. Besides, I did it at first reflash; it´s working now
tks or the support.