johncarlson21/SV04-Marlin-2.1.x #13: Z-axis position is incorrect on display

Piccyman, 2023-04-05:
Go to move screen and the z-axis reading decimal point is out by 10
It reads 1.20 mm instead of 12.0 mm

It just a display error

johncarlson21, 2023-04-21:
yep.. I'm working on this..

GMaassen, 2023-07-26:
Yes, I think this is the same on the "Move" screen, but for all axes. I'm trying to set my off-set using the Ember Prototypes CXC, but I'm one decimal place off.

johncarlson21, 2023-07-26:
I guess I don't know what you mean by this cxc.. what are you trying to do with the move screen? like just trying to find the z?

GMaassen, 2023-07-26:
It's an external camera, developed to find off-sets. Check the website!
What I meant was, in the Settings screen of the Sovol SV04 -> Move. The numbers on this screen seem to be in cm not in mm.

johncarlson21, 2023-07-26:
seems like an extra step, but yes the numbers are off and need to be fixed.. punching in the numbers you put them in correctly, but it shows the wrong numbers.

johncarlson21, 2023-07-26:
I think in your situation the problem is the screen, currently in the latest build and code, I have those * 10 but the screen is looking for * 100, however this caused a problem in the screen as those inputs can only handle so many numbers and it can only be up to like a specific number, which for the Z which can go to 400+ causes the numbers in the input to show some random messed up number. so I've been working on how to fix that in the screen so we can get 2 decimal places, but for the Z this seems not to work correctly. Also X is off once it gets to like over 350 I believe when using 2 decimal places.

LilBub, 2025-01-01:
Hi,
First, you undertook a great challenge with this firmware and I thank you
I'm also using the Ember Prototype XY Calibration camera and it prompts the user to enter the XY position after you align the camera cross hairs with the nozzle.  Its a pretty slick means to calibrate but like others mentioned, the output on the screen is not giving good information,  The calibration asks you to raise the nozzle up 30mm.  After homing, the nozzle is already above the bed, but not reading correctly - showing 1mm when I suspect it means 10?  (X and Y read 15.10 and 12.60 - my logic is that is 151 x 126?  Is there anything we can do to assist in correcting this bug?  I'm willing to contribute to the cause.

johncarlson21, 2025-01-01:
yeah the problem with that is I couldn't get the decimal to come out
correctly, and I'm not working on it anymore.. you are free to fork my repo
and work on it

LilBub, 2025-01-02:
I might try that. I'll download the files you've shared and see what I can do!

LilBub, 2025-03-28:
> yep.. I'm working on this..

@johncarlson21 I reached out the Marlin Firmware group for some help on this.  Someone was kind enough to take a lot at the repository here and provided a potential fix with the screen firmware.

Link is here - https://github.com/MarlinFirmware/Configurations/issues/1135

johncarlson21, 2025-03-28:
So the problem I had was with the screen not really the marlin code. I can't remember anymore. But I'm no longer working on this.

LilBub, 2025-03-28:
Right, I recall you saying that.  I was trying to reach out and help get an answer.  They provided a fix, it appears (check link!)