johncarlson21/SV04-Marlin-2.1.x #26: Extruder 0 (Left) crashes into Extruder 1 when running G29 in Dual Mode

ThankYouSenpai, 2023-04-23:
**Describe the bug**
In Dual Mode, my start gcode includes G29.  When running, Extruder 0 (left) begins in what seems to be position 4/49, near the front right corner.  It then continues probing, going right, until crashing into Extruder 1(right), as if it believes it began at 1/49, or the front left corner.  When running G29 in any other mode this does not happen.

**To Reproduce**
Steps to reproduce the behavior:
Slice model in Cura with G29 in start gcode
Change print mode to dual on touch screen
Start print, confirm mode is correct
Leveling begins in the wrong position.

**Expected behavior**
Slice model in Cura with G29 in start gcode
Change print mode to dual on touch screen
Start print, confirm mode is correct
Leveling begins at front left corner, building the mesh.
Print begins.

**Printer Information**
 Sovol SV04
johncarlson21 v2.1.3.1
Touchscreen v1.2.7

**Additional context**
Add any other context about the problem here.
Original post below:
First let me begin by saying thank you for all the development you have done thus far, you saved this printer for me.

Here is my issue:
In my start gcode with Cura, I run G29 on all SV04 profiles (Single, Copy Mirror, etc). On everything but Dual mode it works fine. In Dual mode, Extruder 0 begins auto leveling at what seems to be position 4 in the regular mesh, so not quite the front right corner, but close. It begins the routine, heading to the right, but it seems that the printer believes it is beginning at the front left corner. This causes it to crash into Extruder 1 (right), then continuing on.

To print, I've removed G29 altogether and it worked beautifully (just finished a 25 hour print). However, I'm worried about not having a mesh active when I do this. Call me paranoid.

Again, G29 works perfectly fine in other modes. Thank you for reading!

[SV04DualGCode.zip](https://github.com/johncarlson21/SV04-Marlin-2.1.x/files/11304079/SV04DualGCode.zip)

ThankYouSenpai, 2023-04-23:
Update:
Slicing with Prusaslicer seems to have fixed it.  I will leave this open for a day or so, but feel free to close this since essentially I no longer have an issue.

johncarlson21, 2023-04-29:
can you show me your gcode from cura so I can see what is happening, and also the prusaslicer gcode so I can compare the start code